XCSoar/XCSoar #1807: XCSOAR IOS Simulator

XCNav, 2025-06-29:
XCSoar versions having the problem
-------------------------------------------------

XCSOAR IOS

XCSoar versions not having the problem
-------------------------------------------------

System information
------------------

IOS 18.5 iPhone XR

Steps to reproduce the behavior
-------------------------------

Try simulator 

Expected behavior
-----------------

Dragging the A/C symbol should start it moving 

Actual behavior
---------------

Info field shows respective G/S changing but A/C symbol does not move. 

Do you have any idea what may have caused this?
-----------------------------------------------

No glue

Do you have an idea how to solve the issue?
-------------------------------------------

WolfgangDrescher, 2025-08-24:
The problem is in `EventQueue::Wait`. Does the simulator work on Android with the current master build? This code fixes the problem on iOS, but it’s probably not the ideal solution.

```cpp
bool
EventQueue::Wait(Event &event) noexcept
{
  if (quit)
    return false;

  while (true) {
    FlushClockCaches();
    if (Generate(event))
      return true;
    
#ifdef __APPLE__
    if (SDL_PollEvent(&event.event))
      return true;

    SDL_Delay(16);
#else
    int result = SDL_WaitEvent(&event.event);
    
    if (result != 0)
      return result > 0;
#endif
  }
}
```

Can I safely put `FlushClockCaches()` inside the loop? I know that @yorickreum refactored this method quite a bit. What do you think about using `SDL_PollEvent` and `SDL_Delay` here?

WolfgangDrescher, 2025-08-24:
I can confirm that reverting commit 1c9ce78 allows the simulator to work again. My iPhone is currently quite hot from charging, so I can’t say whether this has any negative impact on the power consumption. Before opening a PR, maybe @lordfolken can share his thoughts. I still don’t know SDL well enough to determine the best solution for this issue.