XCSoar/XCSoar #1808: IOS: .XCI file seems not to be recognized

XCNav, 2025-06-30:
XCSoar versions having the problem
-------------------------------------------------

XCsoar IOS

XCSoar versions not having the problem
-------------------------------------------------

System information
------------------

IOS 18.5 iPhone XR

Steps to reproduce the behavior
-------------------------------

Download the glb-xci-xcremote-xcnav.xci file in The events tab and restart xcsoar. Install a keyboard and press "M"

Expected behavior
-----------------

With the .xci file installed the xcvario Menu should pop up

Actual behavior
---------------

Nothing happens

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

yorickreum, 2025-08-24:
Thanks for sending me the test device, @XCNav!

Our current evaluation concluded that .xci files probably are indeed working correctly and the underlying problem is that on iOS the app doesn't correctly recognize that a keyboard is connected.

E.g., the quick menu and MC ready / Vario menus can be opened correctly via the XCRemote. However, it does not show the currently selected menu button in blue, as it does under Android if a keyboard input device is connected.

I will investigate this further.

yorickreum, 2025-09-10:
We'd need to (at least) implement `HasCursorKeys()` for iOS in https://github.com/XCSoar/XCSoar/blob/506de14dc0a59aa82f3be256ab7ee3ec17f1387e/src/Asset.hpp#L177C1-L196C2

On Android, this is done but triggering on the first press of any cursor key, see https://github.com/XCSoar/XCSoar/blob/506de14dc0a59aa82f3be256ab7ee3ec17f1387e/src/Android/EventBridge.cpp#L53C1-L68C2

yorickreum, 2025-12-10:
Thanks a lot! I will compile a new version for TestFlight in the next days.

closed 2026-01-09 by yorickreum