XCSoar/XCSoar #1815: Add a BluetoothHelper for iOS devices

WolfgangDrescher, 2025-07-12:
XCSoar version
--------------

XCSoar 7.43 on an iPhone with iOS 18.5

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Currently, it is not possible to configure a Bluetooth device via the NMEA settings on iOS. It would be nice to add a `BluetoothHelper` for iOS devices.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

To download IGC files from the Flarm, you currently need to use a non-iOS device. It would also be helpful to provide GPS data from the Flarm via a Bluetooth adapter to XCSoar, so we don't have to rely on the phone’s internal GPS.

WolfgangDrescher, 2025-07-21:
I did some research on where to start with this. The iOS Simulator actually displays some Bluetooth devices that are paired (but not currently connected) to the Mac. This surprised me, as on a real iPhone, no Bluetooth devices appear in the NMEA device list.

I'm not sure where this list is actually populated. Am I looking in the right place with `src/Dialogs/Device/PortDataField.cpp`, `FillAndroidBluetoothPorts`, and `src/Dialogs/Device/PortPicker.cpp`? I couldn’t find a loop that iterates through all Bluetooth devices, so I'm not sure if I need to dig deeper.

Is my assumption correct that once XCSoar can detect and communicate with Bluetooth devices, and appropriate adapters have been written, it should be able to receive and interpret data from Bluetooth devices and use them for the metrics?

Maybe @yorickreum has some experience with this?

yorickreum, 2025-07-29:
Thanks a lot for your continued efforts!

> I did some research on where to start with this. The iOS Simulator actually displays some Bluetooth devices that are paired (but not currently connected) to the Mac. This surprised me, as on a real iPhone, no Bluetooth devices appear in the NMEA device list.

I assume here the problem is that iPhones just won't connect to devices that are not Bluetooth LE or MFi? The Mac will…

 > I'm not sure where this list is actually populated. Am I looking in the right place with src/Dialogs/Device/PortDataField.cpp, FillAndroidBluetoothPorts, and src/Dialogs/Device/PortPicker.cpp? I couldn’t find a loop that iterates through all Bluetooth devices, so I'm not sure if I need to dig deeper.

You probably figured that out by now.

> Is my assumption correct that once XCSoar can detect and communicate with Bluetooth devices, and appropriate adapters have been written, it should be able to receive and interpret data from Bluetooth devices and use them for the metrics?

I'd think so ... not sure about the details here.

> Maybe @yorickreum has some experience with this?

Not much.

What might be interesting/related: According to https://github.com/XCSoar/XCSoar/discussions/1351#discussioncomment-13609454 the XCRemote connected via BLE seems to work out-of-the-box with the currently published version without problems.

XCNav, 2025-07-30:
The XCRemote basically is a simple BLE or USB Hid device wich acts like a keyboard, sending defined Key characters. The .xci file basically adds an other menu and defines special XCRemote related actions. If you send me a mail to info@xcnav.de I will send you a PCB for free for your development.

XCNav, 2025-07-30:
> > No idea why... To me, it seems like this shouldn't be possible. But I'm not that deep into the codebase yet.
> 
> @XCNav, can you maybe elaborate on how exactly you connected the XCRemote? Does it show up as a NMEA device in the app or did you just connect the iPhone and it identifies as an input device, unrelated to XCSoar? Never had the pleasure to use one, so I'm not sure how you do it...

The XCRemote basically is a simple BLE or USB Hid device wich acts like a keyboard, sending defined Key characters. The .xci file basically adds an other menu and defines special XCRemote related actions. If you send me a mail to info@xcnav.de I will send you a PCB for free for your development.

XCNav, 2025-07-31:
> It seems that the XCRemote is connected via BLE through the iPhone’s Bluetooth settings and is recognized as an external keyboard. In that case, it shouldn't be related to the NMEA device integrations within the XCSoar settings. If some actions of the XCRemote don't work on iOS yet, I'd guess that certain keybindings are missing. However, I'm not yet familiar with where those are defined in the codebase.

The XCRemote is, like you expected, a simple Keyboard. A configuration in the Devices tab is not necessary. It sends partially XCSoar known key commands such as F1 for Quick menu F4 for Flarm screen, ESC, ENTER etc. Those button work on IOS. However a few functions are defined in the .xci file you can download in seetings: config-system-look-language/input-events. Somehow the IOS version does not seem to access the .xci files, so those defined buttons do not work yet.

yorickreum, 2026-08-24:
> Could you try again with [9859111](https://github.com/XCSoar/XCSoar/commit/98591119dc7673db25add917ba8c2be0e9a371b0)?

Now/soon ("external" tests undergo a quick test by Apple) available as Build 114 via TestFlight for you and Markus (@XCNav).

WolfgangDrescher, 2026-08-28:
@XCNav Have you had a chance to test my branch yet, using the TestFlight build Yorick gave you?

XCNav, 2026-08-28:
@WolfgangDrescher @yorickreum Not yet, did not recognize that there is a new release yet. Will test tomorrow and let you know

WolfgangDrescher, 2026-08-29:
Today I tested the same self built BLE adapter in another glider, this time with the FLARM sitting behind an XCVario. It works noticeably better: binary mode came up after 4 pings, the flight list was read, and the progress bar ran to 100 %. Of 48 data frames 38 arrived complete and 10 were cut off in the middle, so the link still loses frame tails, but much less often than in the other glider I have tested so far, which has legacy hardware. Not much is missing for a working flight download. Three things came up, two of which I think are independent of this PR.

## 1. A completed download is discarded when a frame had to be retried

`FlarmDevice::DownloadFlight()` detects the end of a flight by the 0x1A byte at the end of the last packet, read via `AllocatedArray::back()`. That array is reused by the three retries of one `GETIGCDATA` request, and `GrowDiscard()` only ever grows it, so after a retry `back()` returns a stale byte of the longer frame of the failed attempt.

In my log the last packet (173 bytes, 100 %) arrived on the second attempt after the buffer had grown to 463 bytes: end of file missed, one more frame requested, immediate NACK, the whole flight thrown away and fetched again from the start.

```diff
diff --git a/src/Device/Driver/FLARM/Logger.cpp b/src/Device/Driver/FLARM/Logger.cpp
index d66b34a0..96bc78fb 100644
--- a/src/Device/Driver/FLARM/Logger.cpp
+++ b/src/Device/Driver/FLARM/Logger.cpp
@@ -330,13 +330,20 @@ FlarmDevice::DownloadFlight(Path path, OperationEnvironment &env)
     const auto progress = static_cast<unsigned>(data[2]);
     env.SetProgressPosition(std::min(progress, 100u));
 
-    const char last_char = (char)data.back();
-    bool is_last_packet = (last_char == 0x1A);
+    // Read IGC data
+    std::span<const std::byte> payload{data.data() + 3, length};
+
+    /* the last packet ends with 0x1A; look at the end of this frame,
+       not at the end of the buffer: AllocatedArray::GrowDiscard()
+       never shrinks, and the buffer is reused by the retries above, so
+       after a retry its last byte belongs to the longer frame of a
+       failed attempt */
+    const bool is_last_packet = !payload.empty() &&
+      payload.back() == std::byte{0x1A};
     if (is_last_packet)
-      length--;
+      payload = payload.first(payload.size() - 1);
 
-    // Read IGC data
-    os.Write({data.data() + 3, length});
+    os.Write(payload);
 
     if (is_last_packet)
       break;
```

## 2. Lost frames are expensive

`ReceiveEscaped()` waits for the whole frame timeout even when the frame stopped arriving halfway, so every lost frame costs the full timeout, three times over because of the retries. A 6 kB flight took 3.4 minutes, almost all of it waiting. The patch waits at most three seconds for the next chunk, restarted by every chunk that arrives, so a slow but healthy link should be unaffected.

```diff
diff --git a/src/Device/Driver/FLARM/BinaryProtocol.cpp b/src/Device/Driver/FLARM/BinaryProtocol.cpp
index fbdfdfae..a3785748 100644
--- a/src/Device/Driver/FLARM/BinaryProtocol.cpp
+++ b/src/Device/Driver/FLARM/BinaryProtocol.cpp
@@ -119,6 +119,18 @@ ReceiveSomeUnescape(Port &port, std::span<std::byte> dest,
   return p;
 }
 
+/**
+ * Give up on a frame after this much silence, even if the caller
+ * allows more time for the whole frame.  A bridge which drops the
+ * rest of a frame (e.g. a Bluetooth LE adapter with a short transmit
+ * queue) is detected in seconds instead of blocking until the frame
+ * timeout has expired, and the caller can retry that much earlier.
+ * Any link which is still delivering data keeps the frame alive,
+ * because each chunk restarts this timeout.
+ */
+static constexpr std::chrono::steady_clock::duration
+FRAME_IDLE_TIMEOUT = std::chrono::seconds{3};
+
 bool
 FLARM::ReceiveEscaped(Port &port, std::span<std::byte> dest,
                       OperationEnvironment &env,
@@ -131,8 +143,11 @@ FLARM::ReceiveEscaped(Port &port, std::span<std::byte> dest,
   // Receive data byte-by-byte including escaping until buffer is full
   std::byte *p = dest.data(), *end = p + dest.size();
   while (p < end) {
+    const TimeoutClock idle_timeout{std::min(timeout.GetRemainingOrZero(),
+                                             FRAME_IDLE_TIMEOUT)};
+
     p = ReceiveSomeUnescape(port, {p, std::size_t(end - p)},
-                            env, timeout);
+                            env, idle_timeout);
     if (p == nullptr)
       return false;
   }
```

Both patches were written with the help of an LLM from my log, and I do not know the FLARM logger code well enough to judge them. Could someone who does have a look? In particular I am unsure whether the shorter idle timeout could upset other devices, for example one that legitimately pauses in the middle of a frame, and whether the end of file change has side effects I am not seeing. I am happy to drop either of them if they are the wrong approach, the analysis is the part I am confident about.

I'm pinging @Turbo87 @August2111 and @lordfolken as it looks like you have been involved quite a bit in `FLARM/Logger` and `BinaryProtocol`.

## 3. Flight download is unavailable with the XCVario driver

The button is enabled from `DeviceDescriptor::IsLogger()`, which only checks the driver flags, and `xcv_driver` has none. With the FLARM driver the download works, so the XCVario forwards the binary protocol transparently, but then `$PXCV` is lost, so no vario or MacCready sync.

Declaration already handles this case: `CanDeclare()` also accepts `blackboard.IsFLARM(index)`. Options would be to extend `IsLogger()` the same way, or to allow a second driver on the port without requiring `EnablePassThrough()` (which LX needs for mode switching, the XCVario appears not to).

Does anyone have experience with the XCVario routing configuration, in particular whether it keeps sending its own `$PXCV` into the same channel during a binary transfer?

WolfgangDrescher, 2026-08-29:
Also pinging @holmesal, as it looks like you're working on this issue as well. Have a look at my current branch: https://github.com/WolfgangDrescher/XCSoar/tree/ios-bluetooth-claude2

The code is still quite messy at this point. I don't have a FLARM device at home to test against, so I have to head to the airfield at the weekend to debug it properly. Just FYI, so we don't end up duplicating work.

XCNav, 2026-09-12:
> @XCNav Have you had a chance to test my branch yet, using the TestFlight build Yorick gave you?

Sorry for the late answer. There seems no new release so far on test flight. I tested with an ESP32 based Flarm Simulator which can broadcast wether via BT, BLE OR wiFi. No success at the latest available release on BLE on my I pad, whilst 7.46 on Android now also connected successfully to BLE. On the I pad the Flarm was not even visible in BT or BLE. Let me know if you are interested in the farm sim.