XCSoar/XCSoar #1834: iOS: Prevent app restart XCSoar was running in background

WolfgangDrescher, 2025-07-30:
XCSoar versions having the problem
-------------------------------------------------

Unreleased XCSoar 7.44

XCSoar versions not having the problem
-------------------------------------------------

unknown

System information
------------------

iPhone with iOS 18.5

Steps to reproduce the behavior
-------------------------------

1. Open XCSoar and interact with the map
2. Go back to the iPhone home screen (XCSoar is running in the background now)
3. Wait until the iPhone locks the screen (set it to 30s for testing) and maybe wait another minute or so
4. Reopen XCSoar

Expected behavior
-----------------

XCSoar should continue running where you left it.

Actual behavior
---------------

XCSoar crashes, and you have to restart from the fly/sim screen.

Do you have any idea what may have caused this?
-----------------------------------------------

See https://github.com/XCSoar/XCSoar/issues/1693#issuecomment-3066702973

Do you have an idea how to solve the issue?
-------------------------------------------

<details>
  <summary>Click to see my suggestions in `EventLoop::Dispatch`</summary>

```cpp
void
EventLoop::Dispatch(const Event &_event)
{
  const SDL_Event &event = _event.event;

  switch (event.type) {
  case SDL_APP_WILLENTERBACKGROUND:
    LogFormat("[DEBUG] SDL_APP_WILLENTERBACKGROUND");
    glFlush();
    glFinish();
    // appIsInBackground = true;
    break;

  case SDL_APP_DIDENTERFOREGROUND:
    // appIsInBackground = false;
    LogFormat("[DEBUG] SDL_APP_DIDENTERFOREGROUND");
    break;

  case EVENT_CALLBACK: {
    Callback callback = (Callback)event.user.data1;
    callback(event.user.data2);
  } break;

  default:
    if (top_window != nullptr) {
      if (top_window->OnEvent(event) && _event.IsUserInput()) ResetUserIdle();
    }
    break;
  }
}
```

Unfortunately, that had no effect. The `SDL_APP_WILLENTERBACKGROUND` and `SDL_APP_DIDENTERFOREGROUND` events are being properly detected and logged.

</details>

yorickreum, 2025-07-31:
When the app crashes, do we see anything in the console? Like a context loss / failed to make context error?
Can't find anything in the XCSoar log on my device. Might try checking that with the simulator or my phone attached to my Mac...

WolfgangDrescher, 2025-07-31:
<details>
  <summary>Click to open xcsoar-old.log</summary>

```
[2025-07-31T06:53:42Z] Starting XCSoar v7.44-iOS~git#e134130f39
[2025-07-31T06:53:42Z] SDL_GL config: RGB=5/6/5 alpha=0 depth=24 stencil=8
[2025-07-31T06:53:42Z] GL vendor: Apple Inc.
[2025-07-31T06:53:42Z] GL version: OpenGL ES 2.0 Metal - 101
[2025-07-31T06:53:42Z] GL renderer: Apple A18 Pro GPU
[2025-07-31T06:53:42Z] GL extensions: GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_packed_depth_stencil GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_vertex_array_object GL_EXT_blend_minmax GL_EXT_color_buffer_half_float GL_EXT_debug_label GL_EXT_debug_marker GL_EXT_discard_framebuffer GL_EXT_draw_instanced GL_EXT_instanced_arrays GL_EXT_map_buffer_range GL_EXT_occlusion_query_boolean GL_EXT_pvrtc_sRGB GL_EXT_read_format_bgra GL_EXT_separate_shader_objects GL_EXT_shader_framebuffer_fetch GL_EXT_shader_texture_lod GL_EXT_shadow_samplers GL_EXT_sRGB GL_EXT_texture_filter_anisotropic GL_EXT_texture_rg GL_EXT_texture_storage GL_APPLE_clip_distance GL_APPLE_color_buffer_packed_float GL_APPLE_copy_texture_levels GL_APPLE_framebuffer_multisample GL_APPLE_rgb_422 GL_APPLE_sync GL_APPLE_texture_format_BGRA8888 GL_APPLE_texture_max_level GL_APPLE_texture_packed_float GL_IMG_read_format GL_IMG_texture_compression_pvrtc 
[2025-07-31T06:53:42Z] Display dpi=480,480
[2025-07-31T06:53:42Z] OpenGL: npot=false stencil=0x88f0
[2025-07-31T06:53:42Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:53:42Z] Bluetooth is ON
[2025-07-31T06:53:42Z] [DEBUG] SDL_APP_DIDENTERFOREGROUND
[2025-07-31T06:53:42Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] Startup dialog
[2025-07-31T06:54:01Z] Loading profiles
[2025-07-31T06:54:01Z] Loaded profile from /private/var/mobile/Containers/Data/Application/FEC26810-ACF7-41D0-8DBF-8F0D805498DC/Documents/XCSoarData/default.prf
[2025-07-31T06:54:01Z] Loading language file
[2025-07-31T06:54:01Z] Language: loading resource 'de.mo'
[2025-07-31T06:54:01Z] Loaded translations from resource 'de.mo'
[2025-07-31T06:54:01Z] Loading input events file
[2025-07-31T06:54:01Z] LoadTerrain
[2025-07-31T06:54:01Z] Loading Topography File...
[2025-07-31T06:54:01Z] ReadWaypoints
[2025-07-31T06:54:01Z] Failed to read waypoint file: /private/var/mobile/Containers/Data/Application/FEC26810-ACF7-41D0-8DBF-8F0D805498DC/Documents/XCSoarData/user.cup
[2025-07-31T06:54:01Z] Failed to open '/private/var/mobile/Containers/Data/Application/FEC26810-ACF7-41D0-8DBF-8F0D805498DC/Documents/XCSoarData/user.cup': No such file or directory
[2025-07-31T06:54:01Z] Failed to open 'airfields.txt' from ZIP file
[2025-07-31T06:54:01Z] RASP load
[2025-07-31T06:54:01Z] No rasp data file: Failed to open ZIP archive /private/var/mobile/Containers/Data/Application/FEC26810-ACF7-41D0-8DBF-8F0D805498DC/Documents/XCSoarData/xcsoar-rasp.dat
[2025-07-31T06:54:01Z] ReadAirspace
[2025-07-31T06:54:01Z] Register serial devices
[2025-07-31T06:54:01Z] ProgramStarted
[2025-07-31T06:54:01Z] cannot open /private/var/mobile/Containers/Data/Application/FEC26810-ACF7-41D0-8DBF-8F0D805498DC/Documents/XCSoarData/lua/init.lua: No such file or directory
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] Start at terrain center
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] Opening device Eingebautes GPS & Sensoren
[2025-07-31T06:54:01Z] Opening device BLE Sensor: 79BAAE03-55D1-A1E5-C79A-CC6B84E4C158
[2025-07-31T06:54:01Z] Device 79BAAE03-55D1-A1E5-C79A-CC6B84E4C158 not found
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:01Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:02Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:02Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:54:03Z] [DEBUG] Refreshing screen (app is in foreground)
[... cut ...]
[2025-07-31T06:55:24Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:55:24Z] [DEBUG] Refreshing screen (app is in foreground)
[2025-07-31T06:55:25Z] [DEBUG] SDL_APP_WILLENTERBACKGROUND
```

</details>

I'm using an iPhone 16 Pro with iOS 18.5.

Is there another place where the SDL/OpenGL will render the canvas other than `EventLoop::Get`. In a conversation with AI, I was advised to prevent all `gl*` functions using an `app_is_in_background` condition when the app is running in the background. Is there a global place where we could catch this?

yorickreum, 2025-07-31:
Running it on an actual device with the debugger attached it does not crash so far, no matter how often I lock and unlock the device... However, I am seeing a bunch of `Execution of the command buffer was aborted due to an error during execution. Insufficient Permission (to submit GPU work from background) (00000006:kIOGPUCommandBufferCallbackErrorBackgroundExecutionNotPermitted)` being logged.

WolfgangDrescher, 2025-08-01:
> `Execution of the command buffer was aborted due to an error during execution. Insufficient Permission (to submit GPU work from background) (00000006:kIOGPUCommandBufferCallbackErrorBackgroundExecutionNotPermitted)`

Where can I find these logs? Is `log stream --predicate 'process == "XCSoar"'` the good command for this? And can I access these logs when testing XCSoar on an iPhone?

```
<key>UIBackgroundModes</key>
<array>
    <string>processing</string>
    <string>location</string>
    <string>audio</string>
</array>
```

Did not help…

yorickreum, 2025-08-01:
@coderabbitai, what are your suggestions to solve this? Where in the codebase would be the best place to ensure that no OpenGL ES functions are called (the graphics processor is not accessed) while the app is in the background?

yorickreum, 2025-08-01:
Seems somewhat reasonable to me. Just putting a conditional in the main SDL event loop should also work, at least as a quick check.

yorickreum, 2025-08-01:
I think we’ve got it — see: https://github.com/XCSoar/XCSoar/pull/1835.
@WolfgangDrescher, would you mind testing it on your iPhone as well?
Thanks a lot again for bringing up and discussing this issue!

WolfgangDrescher, 2025-08-01:
Yeah, I just tested your `ios-no-rendering-in-background` banch and the app still crashed after waiting some minutes before reopening XCSoar.

WolfgangDrescher, 2025-08-02:
I still managed to crash the app after. Maybe it is device specific? I'm running an iPhone 16 Pro.

yorickreum, 2025-08-02:
Mhh. What do your error logs look like?

For me, the `default 15:42:19.178495+0200 symptomsd CFSM dynamic blacklist: app de.yorickreum.XCSoar denied (background/audio/plan): 1/0/0` completely vanished.

But I still see some `error 15:43:58.957114+0200 symptomsd COSMCtrl applyPolicyDelta unexpected absence of policy on appRecord XCSoar bg time + grace 2025-07-31 15:46:01.739 +0200 now 2025-07-31 15:43:58.956 +0200`.
However iOS didn't seem to kill the app even after the grace period time mentioned in the log has expired.

yorickreum, 2025-08-02:
Are you sure the Info.plist used in the .ipa got updated?
I sometimes have to enforce that by deleting the Info.* files in the output directory

yorickreum, 2025-08-02:
Okay the current version in TestFlight just also crashed on me at resume ... maybe they just don't get killed while connected to a Mac?

@coderabbitai, any insights? Keep it brief and simple and useful, not like your last suggestion.

yorickreum, 2025-08-03:
I added some background location management in https://github.com/XCSoar/XCSoar/pull/1835.

WolfgangDrescher, 2025-08-03:
I tested it a second time, and unfortunately XCSoar crashed again :( . No idea what I did differently this time. Do you have any idea how I can provide you with useful logs?