XCSoar/XCSoar #1836: Smart ring for monitoring vital signs

Blaubart, 2025-08-02:
I've been experimenting a bit with a smart ring from China. It can measure blood oxygen saturation, heart rate, and blood pressure. The measurement results seem reasonably reliable. I can connect it to my Android device via Bluetooth and select it as a BLE sensor in XCSoar. The NavBox heart rate already exists, and my heart rate is displayed there. Unfortunately, blood oxygen saturation isn't available yet, but I think this NavBox would be even more important.

There are two other problems, though. I set the ring to a refresh interval of 10 minutes using an app on my smartphone. Unfortunately, it doesn't allow for any less. For us, however, an update interval of no more than 1 minute would certainly be useful. If no measured values are transmitted, the connection between the ring and XCSoar is lost.

So, a driver would have to be written in which the refresh interval can be set and which maintains the connection permanently, or establishes it once within the interval and queries the value.

I Think that would be a great Feature for XCSoar!!

A ring for monitoring vital signs would have huge advantages for us glider pilots. It's completely unobtrusive and can be worn under a glove when flying in the mountains! Reliable monitoring would therefore be possible!

lordfolken, 2025-08-07:
I like the idea. A new navbox with a different value would be easy. 
What i haven´t found is a blood saturation bluetooth service id.

Blaubart, 2025-08-07:
That sounds very good! I’m on holidays. When I’m back I will see if I could find out how to adjust measurement interval

Blaubart, 2025-08-17:
I found some information at the internet. Measurement interval can only be set to multiples of 5 minutes. But there are some commands for the Android App to connect the ring, start a measurement and close the connection again:

1. Set up a listening
All blood oxygen related data will be called back through CRPBloodOxygenChangeListener.
bleConnection.setBloodOxygenChangeListener(CRPBloodOxygenChangeListener);

2. Start measuring blood oxygen
bleConnection.startMeasureBloodOxygen();

3. Stop measuring blood oxygen
Stop measuring blood oxygen. If the measurement time is too short, there will be no measurement result. The
result is passed through the CRPBloodOxygenChangeListener.onBloodOxygen() callback.
bleConnection.stopMeasureBloodOxygen();

I have no idea if these information are helpful in our case. But if the driver is able to start the connection, measure the value and close connection again, we could monitor Blood Oxygen Saturation every 30, 60, 90s for example.

dspreitz, 2026-08-27:
I had a go at this issue. See here: https://github.com/dspreitz/XCSoar/releases/tag/spo2-test-1
Pls test and report back.

dspreitz, 2026-08-27:
## SpO2 colour coding for the InfoBox — literature review

Follow-up to the test build above. Right now the InfoBox just shows the raw percentage. I would like it to colour the value like a traffic light, so the state is readable at a glance without interpreting the number. To pick defensible thresholds I did a literature search first. Results below, with sources; I have tried to keep what is actually documented separate from my own conclusions.

**The most important finding up front: there is no official aviation SpO2 threshold.** Neither FAA nor EASA nor the German gliding regulations define one. [FAA AC 61-107B](https://www.faa.gov/documentLibrary/media/advisory_circular/ac_61-107b.pdf) explicitly *cautions against* relying on a pulse oximeter as the sole hypoxia indicator: *"Waiting for the percent oxygen saturation levels reported by the pulse oximeter to fall before taking action may result in a level of hypoxia sufficient to cause impairment."* The widely quoted "below 90 %, go on oxygen" is association and expert-author practice, not a standard. That shapes the whole design.

### Proposed defaults

| Colour | Range | Basis |
| --- | --- | --- |
| Green | >= 90 % | [AOPA ASI Safety Alert "Hypoxia"](https://download.aopa.org/asf/Air-Safety-Institute-Safety-Alert_Hypoxia.pdf) ("90 percent or above … the pilot's brain should be receiving enough oxygen to function normally"); [Soaring Rx](https://www.mhoxygen.com/does-your-pulse-oximeter-mean-what-it-says/) ("90+ % OK"); 90 % is also the most commonly recommended ICU alarm limit ([PMC4891882](https://pmc.ncbi.nlm.nih.gov/articles/PMC4891882/)); same value in German/Swiss gliding material ([SG Basel Fricktal](https://sgbaselfricktal.ch/sauerstoff-im-segelflug/), [segelfliegenausbildung.de 2.4](https://segelfliegenausbildung.de/index.php/theoretische-spl-ausbildung/2-menschliches-leistungsvermoegen/2-4-verwendung-von-sauerstoff)). |
| Yellow | 85–89 % | Lower bound is where three independent sources converge: Nonin factory default low alarm 85 %, Aithre Illyrian second warning 85 %, Soaring Rx "85 % or below → don oxygen". The band also covers what is actually measured in unpressurised aircraft at 12,500 ft without oxygen: 86 +/- 4 % ([PMC11794239](https://pmc.ncbi.nlm.nih.gov/articles/PMC11794239/)). |
| Red | < 85 % | As above; also inside the range for which reviews document impairment of complex cognitive tasks ([Frontiers in Physiology 2021](https://www.frontiersin.org/journals/physiology/articles/10.3389/fphys.2021.665821/full)). |
| Red + blinking / warning message | < 80 % | Below 80 % oximeter accuracy is at its worst (FDA Safety Communication 2/2021) *and* the physiological situation is unambiguous — the correct response is oxygen or descent regardless of the exact number. |

Deliberately **not** chosen: yellow at 92 % (the Aithre default). Real measurements in an unpressurised aircraft at 8,000 ft are already 92 +/- 1 % ([PMC11794239](https://pmc.ncbi.nlm.nih.gov/articles/PMC11794239/)), so a 92 % threshold would sit on yellow for much of a normal cross-country flight at 2,500 m and produce alarm fatigue. (My conclusion, based on that source.)

### 1. Physiological thresholds

- Sea level reference for healthy adults: 95–100 % ([AOPA Pilot, "Saturated"](https://www.aopa.org/news-and-media/all-news/2023/october/pilot/flight-md-saturated)).
- Cognition, the most relevant find: *"Complex and novel cognitive task performance may be impaired between 6,500 and 12,000 ft, which typically invoke an SpO2 of 70–90 %"*; *"Simple cognitive task performance … may not deteriorate until below an SpO2 of 65 %"*; experienced Hawk pilots recognised hypoxia only at around 75 % SpO2 ([Frontiers in Physiology 2021](https://www.frontiersin.org/journals/physiology/articles/10.3389/fphys.2021.665821/full)). Caveat: the summary table in that same article contains SpO2 bands that are not physiologically plausible and are internally inconsistent — I would use the running text, not that table.
- Effects appear before SpO2 drops measurably: learning of complex tasks is already slightly impaired at 5,000 ft and "considerably" at 8,000 ft on air; choice reaction times lengthen at 5,000–7,000 ft ([AsMA Position Paper, ASEM 2008;79:433–9](https://asma.org/wp-content/uploads/2008/01/cruising-altitudes.pdf)).
- Counterpoint, useful for calibrating expectations: Muhm et al., n=502, hypobaric chamber, 20 h — mean SpO2 fell only 4.4 percentage points at 8,000 ft versus 650 ft, and this "did not affect … impairment of sensory or psychomotor performance" ([NEJM 2007](https://www.nejm.org/doi/full/10.1056/NEJMoa062770)).

**Could not be substantiated:** a citable SpO2 value "at which unconsciousness occurs". Aerospace medicine uses Time of Useful Consciousness as a function of *altitude*, not saturation, and AC 61-107B warns that "the TUC does not mean the onset of unconsciousness. Impaired performance may be immediate." I did not want to invent a number here.

### 2. Where the 90 % rule actually comes from

- [AOPA Air Safety Institute Safety Alert "Hypoxia"](https://download.aopa.org/asf/Air-Safety-Institute-Safety-Alert_Hypoxia.pdf): "Below 90 percent, mental function begins to deteriorate", plus a recommendation to check every 10–15 minutes.
- Origin of the number: [AVweb / Aeromedix, Dr. Brent Blue, 1999](http://avweb.com/flight-safety/recommended-use-of-pulse-oximetry-in-aviation/) states plainly that "the FAA has not yet provided any official recommendations for the us [... cut ...]
- Reference data supporting 92 % as a normal finding in healthy adults at sea level.

---

Happy to be corrected on any of this, especially by anyone with aeromedical background — @alenz-68, you mentioned a c-med alpha sensor, that would be a good cross-check against these bands in real flight.

*This is a desk literature review compiled with AI assistance; every source is linked so the numbers can be checked. Given it drives a safety-adjacent display, please spot-check before anything gets merged.*

mlep, 2026-08-27:
Quite interesting review. Thank you for sharing it.
Based on ````Green must not read as "all clear."````, for the default thresholds, I would say "if >= 90 %, do not use green (could lead to pilot's overconfidence). keep it black".

dspreitz, 2026-08-27:
> keep it black

I updated the comment accordingly.

dspreitz, 2026-08-27:
## Follow-up: which regulation actually applies to sailplanes, and how stale a reading can get

Three findings that came out of a second literature pass, all relevant to how long an SpO2 reading should stay valid in the InfoBox. Same rules as before: sources linked, my own inferences marked as such.

### 1. For sailplanes it is Part-SAO, not Part-NCO — and Part-SAO has no numbers

This surprised me, and it matters for a gliding project. EASA sailplane operations fall under [Regulation (EU) 2018/1976 Annex II (Part-SAO)](https://www.legislation.gov.uk/eur/2018/1976/annex/II/division/SAILPLANE/adopted), not Part-NCO.

**SAO.OP.150 Use of supplemental oxygen**, in full:

> "The pilot-in-command shall ensure that all persons on board use supplemental oxygen whenever he or she determines that, at the altitude of the intended flight, lack of oxygen might result in impairment of their faculties or harmfully affect them."

**SAO.IDE.115** then only requires storage and dispensing equipment when oxygen is required under SAO.OP.150.

That is the entire rule. **No altitude figure, no duration, no threshold of any kind.** For comparison:

| Rule | Applies to | Altitude / duration |
| --- | --- | --- |
| [14 CFR 91.211](https://www.ecfr.gov/current/title-14/part-91/section-91.211) | US civil aircraft | 12,500–14,000 ft MSL: crew oxygen if that segment exceeds **30 min**; above 14,000 ft: crew oxygen for the entire time; above 15,000 ft: oxygen provided to each occupant |
| [NCO.OP.190](https://regulatorylibrary.caa.co.uk/965-2012/Content/Document%20Structure/07%20NCO/2%20Regs/19630_NCOOP190_Use_of_supplemental_oxygen.htm) | other non-commercial aircraft | 10,000–13,000 ft: crew oxygen beyond **30 min**; above 13,000 ft: all occupants, any duration |
| **SAO.OP.150** | **sailplanes** | **no figures at all — pilot judgement only** |

My reading of this (inference, not a rule): the regulation delegates the entire decision to the pilot, which makes a quantitative cockpit indication more useful here than in aircraft where a hard altitude limit already forces the issue. It also means XCSoar cannot lean on a regulatory number for its thresholds — there is none to lean on.

*Sourcing caveat:* EUR-Lex was not reachable during the search, so the SAO text above is quoted from the UK retained-EU version on legislation.gov.uk. Worth verifying against EUR-Lex before anyone relies on it.

### 2. The FAA's own numbers on how fast performance degrades

From [AIM 8-1-2](https://www.faa.gov/air_traffic/publications/atpubs/aim_html/chap8_section_1.html):

> "other significant effects of altitude hypoxia usually do not occur in the normal healthy pilot below 12,000 feet. From 12,000 to 15,000 feet of altitude, judgment, memory, alertness, coordination and ability to make calculations are impaired […] **In fact, pilot performance can seriously deteriorate within 15 minutes at 15,000 feet.**"

> "**The ability to take corrective and protective action is lost in 20 to 30 minutes at 18,000 feet and 5 to 12 minutes at 20,000 feet**, followed soon thereafter by unconsciousness."

Worth noting for this issue specifically: the SpO2 validity period in the test build is **15 minutes** — the same span in which the FAA says performance seriously deteriorates at 15,000 ft. That is a coincidence of numbers rather than an argument, but it does make 15 minutes hard to defend as a display validity.

Also checked: the TUC tables in [AC 61-107B](https://www.faa.gov/documentLibrary/media/advisory_circular/ac_61-107b.pdf) Fig. 2-3 and [PHAK ch. 17](https://www.faa.gov/sites/faa.gov/files/regulations_policies/handbooks_manuals/aviation/phak/19_phak_ch17.pdf) Fig. 17-1 start at 18,000 and 20,000 ft respectively. **There are no published TUC figures for FL100 or FL130.** (The two FAA tables also disagree at 22,000 ft — 10 min vs 5–10 min — which I could not resolve.)

### 3. Sensor lag — the most practical finding

Before any XCSoar-side expiry even applies, the reading is already old:

- [MacLeod et al., Anaesthesia 2005](https://pubmed.ncbi.nlm.nih.gov/15601275/): desaturation response time of a **finger** pulse oximeter was **130 s** normothermic and **215 s** under mild hypothermia (vasoconstriction). Forehead and ear sensors were unaffected by the cooling; finger vs head p < 0.0001.
- [FAA/CAMI DOT/FAA/AM-15/8](https://www.faa.gov/sites/faa.gov/files/data_research/research/med_humanfacs/oamtechreports/201508.pdf): in a rapid decompression profile, finger SpO2 lagged the forehead sensor by about **15 s** and read on average **19.75 percentage points lower**.

A cold cockpit and a finger clip are the normal case in a glider. On these numbers the displayed value can be **over three minutes behind the pilot's actual state** before XCSoar adds any validity window on top. Two consequences, both my inference: the validity period should be short rather than generous, and ear or forehead sensors (such as the c-med alpha @alenz-68 mentioned) are materially better suited than finger clips for this use.

Recovery is quick in the other direction: SpO2 returned to baseline **within 3 min** after descent in a chamber study at a 10–15 m/s descent rate ([PMC12894062](https://www.ncbi.nlm.nih.gov/pmc/articles/PMC12894062/)).

### 4. Explicit dead ends, so nobody repeats the search

[... cut ...] like the answer to this question and is not one.

---

A concrete proposal for the validity period still needs one more piece: what real devices actually use as a measurement interval, and whether there is precedent in cockpit systems for displaying a reading's *age* rather than expiring it outright — the FIS-B and NEXRAD age-annotation problem being the obvious comparison. I will follow up with that.

*Literature review compiled with AI assistance; sources linked so the numbers can be checked.*

dspreitz, 2026-08-27:
## Changes to the SpO2 InfoBox, and one caveat that needs flagging

Following the two literature passes above, I have changed two things in the test branch.

**1. Validity reduced from 15 minutes to 3 minutes** (`src/NMEA/Info.cpp`).

The original 15 minutes were chosen on the assumption that some rings only measure every 5–10 minutes. That assumption is still unsourced — I could not find a manufacturer figure for the Colmi rings or the c-med alpha, and Oura turns out to measure SpO2 only during sleep, which rules it out for cockpit use anyway. Meanwhile the evidence against a long window is concrete: a finger pulse oximeter's own desaturation response time is **130 s normothermic and 215 s with cold fingers** ([MacLeod et al., Anaesthesia 2005](https://pubmed.ncbi.nlm.nih.gov/15601275/)), so a generous display window stacks staleness on top of staleness. Three minutes is in the same order as the sensor's own lag and shorter than the time a glider needs to reach a materially different altitude band.

The consequence is deliberate and worth stating plainly: **a sensor that genuinely reports only every 5–10 minutes will show "invalid" between measurements.** I think that is the correct message rather than a defect — a device with that duty cycle cannot support a hypoxia indication in a cockpit, and making it look as though it can is the larger risk. If someone has a ring with a documented measurement interval, that data would be very welcome, since the whole 15-minute rationale rested on it.

**2. The InfoBox now shows the age of the reading** once it is older than 30 s, in the comment line.

This follows established cockpit practice rather than medical practice. Medical devices blank a stale reading fast — the [Nonin PalmSAT 2500 manual](https://www.nonin.com/wp-content/uploads/2018/09/Operators-Manual-2500-1.pdf) specifies that on inadequate signal the last values "freeze for 10 seconds and are then replaced with dashes". Aviation systems instead keep showing the data with an age annotation. Given that a glider pilot may have no second source at all, showing "88 % / 2 min" seems more useful than showing nothing.

## The caveat: this is the age of reception, not of measurement

This is the part that needs flagging before anyone relies on it.

`DeviceDescriptor::OnBloodOxygenSensor()` stamps the value with `basic.clock`, i.e. the moment XCSoar received the BLE notification. The PLX Spot-check Measurement characteristic (0x2A5E) has an **optional timestamp field** that the driver does not currently parse. For a sensor that buffers internally and transmits late, the displayed age is therefore **smaller than the true age of the measurement**.

That is not a hypothetical failure mode; it is a documented fatal one in a neighbouring domain. [NTSB Safety Alert SA-017](https://www.ntsb.gov/Advocacy/safety-alerts/Documents/SA-017.pdf) exists precisely because in-cockpit NEXRAD displays understate the age of what they show:

> "Weather conditions depicted on the mosaic image will ALWAYS be older than the age indicated on the display."

> "in extreme latency and mosaic-creation scenarios, the actual age of the oldest NEXRAD data in the mosaic can EXCEED the age indication in the cockpit by 15 to 20 minutes."

Two accidents are cited in that alert where the displayed age was roughly a minute while the actual data was 5 to 8 minutes old. The lesson is not "don't show age" — it is "don't show an age that is smaller than the truth".

For now the 3-minute validity bounds the exposure, and the code comment says explicitly what the number means. Parsing the spot-check timestamp would be the proper fix, and it has some non-obvious problems of its own (the Continuous Measurement characteristic has no timestamp field at all; the spot-check timestamp is a wall-clock date-time in an unspecified time zone; and the flags carry a "Device Clock is Not Set" bit precisely because device clocks often are not). I have looked at what that would involve and can write it up separately if there is interest — I did not want to build it into this change.

Also opened: #2970, on the palette colour this InfoBox needs for its caution band.

mlep, 2026-08-28:
About
 ````
SAO.OP.150 Use of supplemental oxygen, in full:

    "The pilot-in-command shall ensure that all persons on board use supplemental oxygen whenever he or she determines that, at the altitude of the intended flight, lack of oxygen might result in impairment of their faculties or harmfully affect them."
````
This is not in full: this is only the first of the two paragraphs of this section (source: https://www.easa.europa.eu/sites/default/files/dfu/Sailplane%20Rule%20Book.pdf). It continues as:
````
GENERAL
When the pilot-in-command cannot determine how the lack of oxygen might affect the persons on
board, he or she should ensure that all occupants use supplemental oxygen for any period when the
pressure altitude is above 10 000 ft.
````

So, the recap in the table "SAO.OP.150 	sailplanes 	no figures at all — pilot judgement only" should be altered. For sailplane, there is a clear figure: ````any period above 10 000 ft````.

_BTW: who could this pilot-in-command be who can determine how the lack of oxygen might affect the persons on board?_

alenz-68, 2026-08-31:
I performed a small test on the code of @dspreitz.  I test with a cosinuss C-med alpha in ear sensor. My findings are:

Connection procedure for me was as described below, as the sensor is a BLE device:

Once the device is powered on, it will show up in the device list. In the case of my sensor it is the cmed device at the end of the list:

[image]

Then make sure to set "_Motor Typ_" to "_None_", I think, the reference to motor typ is because, BLE was mainly used for engine monitoring devices so far. The UI might be improved for other types of BLE devices like vital sign sensors.

[image]

After pressing OK, i first get an exception with GATT connection error. It takes a few seconds, then the device shows up as connected.

[image]

Both values show up in their apporpriate infoboxes. Data looks reasonable, update rate is about 1 Hz, but i guess that depends upon the notification rate of the device.

[image]

A possible isssue for all BLE devices is that i cannot deactivate the device. When i select "deactivated" in the connection manager, it is still reported as connected.

Regarding the discussion on possible warning thresholds the literature review is quite interesting. But it also shows, that there are no agreed warning levels for SpO2 measurements.[Pilot Medical Monitoring: State of the Science Review on Identification of Pilot Incapacitation](https://rosap.ntl.bts.gov/view/dot/67587) mentiones ranges of no concern (94% or greater), low concern (87%-94%) and high concern (below 87%).
But you have to be aware, that pilots want to avoid incapacitation effects caused by hypoxia, which is the oxygenation of the (brain) tissue. Typical sensors measure just blood oxygenation levels, with some limited accuracy.
So i would opt for just displaying the measurements (and recording within the IGC file) and not reason about it but leave that to the educated pilot.

dspreitz, 2026-08-31:
I created a new build, which stacks some individual commits for the three addressed issues. @alenz-68 can you please test again and report back?
[https://github.com/dspreitz/XCSoar/actions/runs/33410972877#artifacts](https://github.com/dspreitz/XCSoar/actions/runs/33410972877#artifacts
)

alenz-68, 2026-09-01:
Test resuts for latest build:

Data for both heartrate and oxygen saturation (SpO2) is read correctly from device as expected but connection handling could still be improved.
I tested on Android smartphone with [cosinuss cmed-alpha](https://www.cosinuss.com/de/produkte/im-ohr-sensoren/c-med-alpha/), which reports heartrate and SpO2 as a BLE device using the standard GATT protocols.

- starting XCSoar with no device selected, under "Config>Devices" i see
```
A: Disabled
Disabled
```
- I select port A and Edit, choose cmed/BLE sensor from list
- once port is selected, set _Engine Type_ to _None_ . In some future release it might make sense, to modify the UI, chaning engine Type to protocol/driver with an explicit option for GATT profiles.
- listed devices now show:
```
A: BLE sensor: cmed
java.io.IOException: Bluetooth GATT connect failed
```
and after approximately 5 seconds:
```
A: BLE sensor: cmed
Connected
```
- correct data is now displayed in the Infoboxes at the device update rate of approximately 1 sec
- now under "Config>Devices" i edit _A: BLE sensor: cmed_ and set the Port to _Disabled_
- Though port A should now be disabled, it is still connected and data is still updated in infoboxes
```
A: Disabled
Connected
```
- Trying connect the device on Port A under "Config>Devices" is not possible, as the device no longer shows up in the list (as it still is connected, even if not visble)
- i quit XCSoar and launch again, connection is not reestablished so i did the same connect procedure
- quit again, with device configured
- launch XCSoar again, connection is established and data is written to Infoboxes without need for any configuration
- Checking the device configuration it now shows:
```
A: BLE sensor: EB:31:DB:D3:C9:29
Connected
```
Device name is not resolved, instead just its ID shows up. This behavior might not be specific to the vital signs sensor but could be specific to all BLE devices.
- switching the device of is recognized as "GATT disconnected". Switching the device on again requires a manual Reconnect.

For me the current implementation is acceptable. If we would like to support more BLE devices/sensors (i could think of outside air temperature or gear/flap sensors) BLE connection handling should be improved.  

Personally, i would like to see the vital signs data included in the IGC file, which might help to understand e.g. which situations caused stress to the body. Technically this should be possible using extensions of the B-Record, defined in an I-Record. But this might be more a regulatory issue, as vital signs are sensitive data, that fall under data protection regulations. Any extension to the IGC log would also have to be compatible with scoring tools like WeGlide.

dspreitz, 2026-09-01:
Hi @alenz-68, thanks for the detailed test report — and sorry, the build you tested was faulty.

The fix for `java.io.IOException: Bluetooth GATT connect failed` was not actually in `spo2-test-2`. When I assembled the branch, only half of that change made it in, so the broken check was still in place and the error still appeared on every connect. Your report was right, the build was wrong. `spo2-test-3` had the same defect.

I have rebuilt the branch from scratch and checked it commit by commit: **[spo2-test-4](https://github.com/dspreitz/XCSoar/releases/tag/spo2-test-4)**.

Two things changed since the build you tested:

- **The connect error should be gone.** The constructor checked the GATT handle immediately after handing the connect to the Android main thread, so on Android 6 and newer it always found it unset. That check now only runs on the branch that connects synchronously.
- **A disabled device should now actually release the sensor.** `close()` released the GATT client but never told the peripheral to drop the link, so the connection stayed up — which is why the sensor did not reappear in the scan list. It now disconnects first and waits for the disconnected state before releasing the client, the same sequence `BleSerialPort` has always used.

If you have time for another round, the two things worth watching are whether the error still appears while connecting, and whether **Edit → Port → Disabled** stops the InfoBox values and lets you pick the sensor again in a fresh scan.

Your other four points are noted and deliberately not addressed in this build: the MAC address instead of the device name after a restart, the missing reconnect after switching the sensor off and on again, *Engine Type* as the label for a vital-signs sensor, and vital signs in the IGC file. The first two are BLE connection handling and belong in the same corner of the code; the label is a UI question; the IGC part I would keep separate, since as you say it touches data protection and compatibility with scoring tools.

alenz-68, 2026-09-02:
I tried to connect the device following the same procedure as before. Connection could not be established. The "Config > Devices" list shows:
_A: BLE sensor: cmed
Not connected_ 

i also checked the xcsoar.log, it just includes the message:
_[...] Opening device: BLE sensor: cmed_

One remark on the test yesterday. When i tried to disable the device, i edited the port and selected disabled, i did not use the disable button.

dspreitz, 2026-09-02:
@alenz-68 thank you — and sorry, that is a regression from my change, not your setup.

*Not connected* is what XCSoar shows when the sensor reported **neither** success nor failure. So the connect was started and nothing ever came back. Two different causes produce exactly that picture, and they need different fixes, so before I build anything I would like to narrow it down. Neither test needs anything but the phone.

**Test 1 — does Reconnect help?**

1. Force-stop XCS-test and start it again, so nothing is left over from the failed attempt.
2. Go to **Config → Devices**. Device A should show *Not connected*.
3. Leave it alone for about a minute without touching anything — does it change on its own?
4. Then select the row for device A and press **Reconnect**. Give it about 10 seconds.
5. If nothing happens, press **Reconnect** a second time.

What I need to know: does it stay *Not connected*, does it connect, or does an error message appear?

**Test 2 — what does the phone think?**

While XCSoar still shows *Not connected*, check the c-med alpha in the Android Bluetooth settings or in the cosinuss app: is the sensor connected there?

That single question separates the two causes. If the phone says connected while XCSoar does not notice, then I am discarding a callback I should not discard. If the phone also says not connected, then Android never answers the connect attempt — and the code has no timeout anywhere, so it waits forever.

For the record on what changed: the earlier builds threw an exception on every connect. That exception was wrong, but it was also the only thing that produced any outcome when Android does not call back, and the connection you saw appear a few seconds later was made by the object created just before it. Removing it left the retry with nothing to trigger it. Whatever these tests show, I will add a proper connect timeout — waiting indefinitely with no feedback is wrong either way.

alenz-68, 2026-09-02:
Test 1:
Config → Devices. Device A showing _Not connected_
waited a couple of minutes, nothing changed
Selected Row for device A, pressed **Reconnect** -> no change
pressed **Reconnect** second time -> not change, still reporting _Not connected_, no error message
each **Reconnect** attempt is logged as Opening device: BLE sensor: cmed in xcsoar.log

Test 2:
Still showing _Not connected_ cmed does not appear in connected bluetooth devices (just as available device). It is also not visible in the cosinuss app. 
The phones bluetooth configuration will not report a connected BLE device, as it is not bonded.

For debugging Bluetooth and BLE i use the App "nRF Connect". This app shows no connection between the test build and the device while it does for the 7.46 development build. It should also be possible to use the nRF app as an BLE device emulator.

dspreitz, 2026-09-02:
@alenz-68 that nRF Connect observation settled a lot — thank you. If nRF sees no link at all, then nothing is being connected in the first place, so it is not a callback I am discarding somewhere. That rules out one of my two suspects.

Three commits separate the build that worked for you (`spo2-test-2`) from the one that does not (`spo2-test-4`), and one of them is now excluded. To decide between the remaining two, here is a build with exactly one of them removed and nothing else changed:

**[spo2-test-5](https://github.com/dspreitz/XCSoar/releases/tag/spo2-test-5)**

Only one thing needs testing: **does it connect?**

1. Install alongside as usual — package `org.xcsoar.testing`, app **XCS-test**.
2. **Config → Devices**, port A, *BLE sensor*, pick the c-med alpha, *Engine Type* → *None*.
3. Connects or not. That is the whole answer.

If it connects, the cause is a guard I added that suppresses a connect still queued on the Android main thread. If it does not, the cause is that I removed the spurious exception from the constructor — which would mean the connection you have been getting all along was a side effect of XCSoar treating the open as failed, and that the code never really connected on purpose.

Please ignore everything else in this build. The `java.io.IOException: Bluetooth GATT connect failed` message may or may not appear, and a device disabled through **Edit → Port → Disabled** still keeps the sensor connected — that was the removed commit's job. Neither is meaningful here.

alenz-68, 2026-09-02:
I can confirm a connection. Device changes from "_Not connected_" to "_No data_".

dspreitz, 2026-09-02:
That answers it — thank you, that was exactly the test I needed.

`spo2-test-5` connecting pins the failure on one commit, and within it on one thing: a guard I had added that suppressed a connect still queued on the Android main thread. It turns out that queued connect is not incidental, it is how a BLE sensor gets connected here at all. The connect is posted on purpose so that it runs *after* a `close()` that arrives before the connection is established — there is a comment in the constructor saying so, which I should have taken more seriously. Suppressing it meant the sensor never connected.

The guard is gone. What remains is the part that was actually wanted: **[spo2-test-6](https://github.com/dspreitz/XCSoar/releases/tag/spo2-test-6)**

`close()` now calls `disconnect()` first, waits up to 500 ms for the disconnected state and only then releases the client. The connect path itself is character-for-character identical to `spo2-test-5`, the build that connected for you, so the regression cannot come back.

Two things worth testing, in this order:

1. Does it connect, and does the **SpO2** InfoBox show a plausible value against the sensor's own app?
2. Then **Edit → Port → Disabled**. The list should show *Disabled*, the InfoBox values should stop, and the sensor should be selectable again in a fresh scan. That is the part that is new — it is what the connect problem was blocking all along.

One thing I have not fixed and want to name rather than leave you to find it: there is still no timeout on a connect attempt. If Android never answers, XCSoar waits indefinitely with no feedback, which is what you were looking at with `spo2-test-4`. That belongs in a separate change and I will get to it.

Your other four points — MAC address instead of the device name, no reconnect after switching the sensor off and on, the *Engine Type* label, and vital signs in the IGC file — are still open and untouched.

dspreitz, 2026-09-02:
@alenz-68 I found the root cause, and it explains several things in this thread at once — including two of your reports that contradicted each other.

**XCSoar closed every Android sensor immediately after opening it.** `DeviceDescriptor::OpenBluetoothSensor()` wraps the value the factory returns in a `GlobalCloseable`. That value was a `LocalCloseable`, and `GlobalRef` takes its source by const reference — so the temporary survives the construction, and its destructor calls `close()` on the object that was just opened.

For a BLE sensor that runs `safeDestruct.beginShutdown()`, after which `SafeDestruct.increment()` returns false permanently. Everything the sensor reports back sits behind it: no measurement is ever delivered, no error is ever shown, the connect retry silently does nothing — and the later real `close()` throws out of `beginShutdown()`, so `gatt.close()` is never reached and the GATT client is leaked. That last one is your "cannot deactivate it, and it does not come back in the scan".

It stayed hidden because the constructor posts the GATT connect to the Android main thread, so the connect runs *after* that premature close and a link is still established. Only what travels back through `SafeDestruct` was lost.

This is why your two reports never fitted together: `spo2-test-2` still threw in the constructor, so the temporary never existed, no `close()` happened, and you read valid heart rate and SpO2. `spo2-test-5`, with that throw removed, opened cleanly — and therefore closed itself, connected, and showed *No data*. Both were correct observations of the same defect from opposite sides.

**[spo2-test-7](https://github.com/dspreitz/XCSoar/releases/tag/spo2-test-7)** has the fix.

What is worth testing, in this order:

1. Does it connect, and does the **SpO2** InfoBox show a plausible value against the sensor's own app?
2. Then **Edit → Port → Disabled**: does the list show *Disabled*, do the values stop, and can you pick the sensor again in a fresh scan?

Point 2 matters more than usual. I had written a separate change that makes `close()` disconnect before releasing the client, on the assumption that this was the cause of the "stays connected" symptom. It now looks like the real cause is the one above, so I deliberately left that change **out** of this build — if the sensor is released cleanly now, it is not needed at all. If it still hangs on, tell me and I will bring it back with a real justification instead of a guess.

Validated here on a Pixel 3a (Android 12) against a device that cannot answer a GATT connect: three connect attempts, three GATT clients properly released, and the error reported to XCSoar. Before the fix that was one attempt, no release and no error. I still have no BLE pulse oximeter, so the SpO2 path itself remains untested on hardware.

Your other points are still open and untouched: MAC instead of the device name after a restart (analysed — XCSoar stores no name and asks Android, which only knows one for a bonded or recently scanned device), no automatic reconnect, the *Engine Type* label, and vital signs in the IGC file. There is also still no timeout on a connect attempt.

alenz-68, 2026-09-03:
Here my test results for latest build, a bit more than requested:

With the usual connection procedure i get the following output for the device on Port A: 
_A: BLE sensor: cmed_
_Not connected_  (showing up for a very short time)
-> _No data_        (a few seconds, until measurements are available)
-> _Connected_

Readings in infoboxes are fine, both for heartrate and SpO2. Next i edited the port to disabled, output as expected:
_A: Disabled_
_Disabled_
-> nothing written to infobox

Connected the device again, then using the **Disable** button results in:
_A: BLE sensor: cmed
Disabled_
-> nothing written to infobox

Enable again:
-> _No data_
-> _connected_ (correct readings in infoboxes)

Switching device off:
_A: BLE sensor: cmed
GATT connection failed (status 19)_

Switching device on:
_A: BLE sensor: cmed
GATT connection failed (status 19)_

**Reconnect** button (required from above state):
_Not connected_
-> _No data_
-> _Connected_

From device witched off (_Gatt connection failed_), pressing **Reconnect**:
_A: BLE sensor: cmed
Not connected_

device switched on:
_A: BLE sensor: cmed
No Data -> Connected_ (No **Reconnect** required)

Restart XCSoar with device configured while device running: connection is established and data is written to infoboxes.

Restart XCSoar with device switched off: Connection is established when device is switched on.

Temporary connection loss (moving out of range):
_GATT connection failed (status 8)_  (requires **Reconnect**)

current implementation works quite well, though there may be other testcases to be performed:
- test with device, that just provides heartrate and not both values
- other unexpected disconnects (e.g. devices powers down, due to low battery)
- other connection failures (connection timeout, GATT incompatible, maximum number of clients exceeded,...)
- do other BLE devices (engine sensor) still work?

lordfolken, 2026-09-22:
the gatt issues should be fixed in master and v7.45.3. General BLE sensors work and even report battery level.

closed 2026-09-22 by lordfolken