XCSoar/XCSoar #1852: iOS: App crashing when resuming from background

yorickreum, 2025-08-24:
The app still sometimes crashes when resuming from background / a locked iPhone.

XCSoar versions having the problem
-------------------------------------------------

Current in master / TestFlight.

XCSoar versions not having the problem
-------------------------------------------------

Unaware. However, we tried addressing it before in #1835. 
Since then the app now completely closes instead of just renouncing to the FLY/SIM screen.
TestFlight also reports significantly more app crashes since the patch.

System information
------------------

iPhone 13 Pro and others

Steps to reproduce the behavior
-------------------------------

Run the app in Fly or Simulator mode, lock the phone, unlock it again. App sometimes crashes then.

Expected behavior
-----------------

App should stay alive.

Actual behavior
---------------

App crashes.

Do you have any idea what may have caused this?
-----------------------------------------------

-

Do you have an idea how to solve the issue?
-------------------------------------------

Background location access or background GPU access maybe need to be handled more carefully...

lordfolken, 2026-02-01:
I saw this with the current IOS build from apple store.

lordfolken, 2026-04-20:
It typically creates a xcsoar-old.log. Does that list a normal exit?

lordfolken, 2026-09-29:
Triage: labeling this `bug` and taking it off needs-triage. Not closing, and not a 7.45 blocker — the June note still stands that it is infrequent and maybe simulator-only.

It is still a resume-from-background crash on current TestFlight (iPhone 13 Pro and others). #1835 changed the failure from dropping back to the FLY/SIM screen into a full exit. Pointer: the iOS foreground path after that change, not a new UI bug.