XCSoar/XCSoar #1859: iOS version file transfer

DanD222, 2025-08-31:
With the iOS version coming along nicely, it might be worth thinking about how to get additional files, e.g. a club’s or a competition’s waypoints or airspaces, into iOS XCSoar.

I believe this could be done through a Share Extension?

I was alternatively thinking of perhaps an area in xcsoar-data-content where people could upload their files and they’d be automatically deleted after a given period of time, but the Share Extension is probably a better way to do this.

WolfgangDrescher, 2025-08-31:
You can access XCSoar files on iOS via:

```
Files App -> On my iPhone -> XCSoar -> XCSoarData

```

Can’t you just copy the files there manually?

DanD222, 2025-08-31:
Doh! :-/ 

I can; thanks.

closed 2025-08-31 by DanD222

TomOcker, 2026-09-10:
Hello DanD222,
I only found "Apple-Devices" as App on the Windows-PC for dropping files on the installed app "XCSoar". There is a directory named XCSoarData as on Android devices. But I can't copy my extra Airspace-file into this directory. Is the "Files-App" another progrram? 
Thanks, Thomas

WolfgangDrescher, 2026-09-10:
It's a native app by Apple that comes preinstalled on every iPhone and iPad. It's simply called "Files" and should look like this:

[image]
<br>
[image]