XCSoar/XCSoar #1873: XCSoar 7.44 Linux Raspberry PI4 crach when cross start line Task

filosy, 2025-09-07:
XCSoar versions having the problem
-------------------------------------------------

XCSoar 7.44 Linux  with RPI 4 8GB

XCSoar versions not having the problem
-------------------------------------------------

I have not tested other version

System information
------------------

I've been using XCS with a Raspberry PI4 without any major problems, until I tried running a task and noticed that when I cross the start line, XCS resets, and I have to restart it from the dialog box. I also tried replaying the flight with the Replay function, and the same part of the flight with the task inserted resets, but without the task there are no problems.

Any ideas?

How can I help you solve the problem?

Expected behavior
-----------------

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2025-09-07:
Logs (igc,nmea,xcsoar application log), Task file? 
are you flying with a debug build (red)? That one has asserts which will trigger an exit if something is fishy.

lordfolken, 2025-09-07:
In those logs i don't see a crash, but normal program termination. 

[2025-09-07T10:06:58Z] Entering shutdown...
[2025-09-07T10:08:40Z] Entering shutdown...

Which of the logs files have you replayed?
The task file in the logs is very wierd. Its binary and doesn´t contain text.

lordfolken, 2025-09-07:
```
hexdump .tsk 
0000000 ffff ffff ffff ffff ffff ffff ffff ffff
*
00005f0 ffff ffff ffff ffff                    
00005f8
```

lordfolken, 2025-09-08:
Hmm could it be a sound problem?
When crossing the startline you usually have a sound being played with the notification. 

Indicative that that is happening is this in the log.
[2025-09-08T09:52:29Z] Using ALSA PCM device "default" (use environment variable ALSA_DEVICE to override)
[2025-09-08T09:52:29Z] Using ALSA PCM latency 100000 μs (use environment variable ALSA_LATENCY to override)

There is no other indication of a crash in the log. 

Can you launch xcsoar from a bash shell and do the same? Then any segfault or similar should be evident.

lordfolken, 2025-09-08:
I'm stupid. On the image the assert failure is visible:

Basically its tripping the assert in this function:
https://github.com/XCSoar/XCSoar/blob/506de14dc0a59aa82f3be256ab7ee3ec17f1387e/src/Engine/Task/Solvers/TaskSolution.cpp#L58-L72

It thinks that the ac.location.isValid is not true. The question is why.

lordfolken, 2025-09-08:
@coderabbitai can you search for why assert(ac.location.IsValid()); could be false?

lordfolken, 2025-11-02:
Happens on windows as well 

[image]