XCSoar/XCSoar #1875: AirGlide (CAI302) task declaration fails

HoeckDK, 2025-09-07:
XCSoar versions having the problem
-------------------------------------------------

Fairly new. Sorry, do not know the actual one. Anyway it looks like the driver has not been changed for a couple of years. 

XCSoar versions not having the problem
-------------------------------------------------

OV 22304. 

[image]

System information
------------------

CM4 board that replaces cubie board

Steps to reproduce the behavior
-------------------------------

Declaration of task to AirGlide, configured for CAI302 mode fails to program the task in AirGlide. 

Expected behavior
-----------------

Task programmed in AirGlide

Actual behavior
---------------

XCsoar complains that task declaration failed, and task did not upload to AirGlide.
* I think that maybe in previous version it also reported that declaration failed, but it still had the task in AirGlide.

Do you have any idea what may have caused this?
-----------------------------------------------

CAI302 driver must have changed. 

Do you have an idea how to solve the issue?
-------------------------------------------

Maybe revert the driver?

Note: I tried to create a debug of the process, but I see no file created. Should i wait for the full 10 mins, before the debug file is written?

lordfolken, 2025-09-07:
The xcsoar.log contains the debug communication if you select debug on a particular device. 

What would be good:
The Task file that you tried to declare
The xcsoar.log file with debug enabled.
The nmea file if you have it. 

And then "someone" will look at it. hopefully.

HoeckDK, 2025-09-07:
[xcsoar.log](https://github.com/user-attachments/files/22198316/xcsoar.log)

Here is the log file. I think it is valid, even if the timestamps look off. The IGC file looks to have the right date it its name.

lordfolken, 2025-09-07:
[2025-08-15T20:22:52Z] Opening device /dev/ttyAMA3
[2025-08-15T20:22:52Z] Opening device /dev/ttyAMA2
[2025-08-15T20:22:52Z] Opening device /dev/ttyAMA0
[2025-08-15T20:22:52Z] Opening device UDP port 2000
[2025-08-15T20:22:52Z] Failed to set the system clock from GPS: Automatic time synchronization is enabled
[2025-08-15T20:22:53Z] OpenFLARMDetails
[2025-08-15T20:22:53Z] Failed to open '/home/root/.xcsoar/xcsoar-flarm.txt': No such file or directory
[2025-08-15T20:22:53Z] 14673 FLARMnet ids found
[2025-08-15T20:22:54Z] EnableNMEA() failed: Port read timeout

Which port is which device here? 

In the middle of the declaration there is a timeout which resets the communciation:

[2025-08-16T02:09:17Z] R 0000  31 2c 30 2e                                       1,0.
[2025-08-16T02:09:17Z] WaitRead 990946
[2025-08-16T02:09:17Z] WaitRead 990946 = timeout
[2025-08-16T02:09:17Z] EnableNMEA() failed: Port read timeout
[2025-08-16T02:09:17Z] StartRxThread
[2025-08-16T02:09:23Z] StopRxThread
[2025-08-16T02:09:23Z] Flush
[2025-08-16T02:09:23Z] Write(1)
[2025-08-16T02:09:23Z] Write(1)=1

HoeckDK, 2025-09-07:
> [2025-08-15T20:22:52Z] Opening device /dev/ttyAMA3 [2025-08-15T20:22:52Z] Opening device /dev/ttyAMA2 [2025-08-15T20:22:52Z] Opening device /dev/ttyAMA0 [2025-08-15T20:22:52Z] Opening device UDP [...]

I’m prett

closed 2025-09-14 by HoeckDK