XCSoar/XCSoar #1880: SkyLines - tracking frequency

PMRix, 2025-09-13:
XCSoar version
--------------
7.43, but actually all

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Would it be possible to add a possibility of SkyLines tracking frequency change?

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Currently positions of tracked friends are updated every 60 s and, AFAIR with a 3-minute delay. It would be a nice feature to be able to change it, downto e.g. 10 or even 5 s and remove this delay. Especially for shorter distances errors can be quite bit.

TIA

PMRix, 2025-09-17:
Hi,

THX for the reaction. The issue I described is formulated in the confusing way - sorry for this: we - my friend we track each other with and me - are aware of both settings you mentioned, but it seems to us that the tracking does not behave as configured.

We could observe updates every 1 minute and a delay of 3 minutes despite of the most "aggresive" settings of "1 s" and "no delay". We have such settings now and have had also previously - we double-checked it after your reply and reran a short test with similar results.

I mean tracking of a friend's position in the XCsoar and not on the SkyLines website (the "yellow face").

I assumed that apart from these settings there was some "hidden magic optimizing mechanism" implemented (either in XCSoar or SkyLines) that resulted in slower position refreshes.

Can you confirm that in your case the tracking performs exactly as configured - frequently and without a delay? We observe a behavior different from the configured one.

lordfolken, 2026-05-14:
closing as stale.

closed 2026-05-14 by lordfolken