XCSoar/XCSoar #1881: SkyLines - tracking - adding a vector of tracked object's speed

PMRix, 2025-09-13:
XCSoar version
--------------
7.43, but actually all

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Would it be possible to provide tracked objets with some information about their speed (e.g. an arrow with a speed value)?

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
This would help in approaching tracked friends.

PMRix, 2026-11-09:
Follow-up:

And what about LiveTrack24? It had worked for me very good for a long time but suddenly stopped to record live tracks from XCSoar in ca. April-May 2026 :(

PMRix, 2026-11-10:
8-O  Was there any release note etc. regarding this?

Is there any work in progress there?
Can you approximate any time horizon for reenabling of thee LT24 livetracking?

TIA

lordfolken, 2026-11-10:
This is not a business. This is a volunteer driven project, no one earns money here. 
Easiest way is to start implementing and especially test yourself. There is enough LLM instructions that can generate the code in a fashion that will pass the project. 

Otherwise i can put it on the roadmap https://github.com/XCSoar/XCSoar/milestones

PMRix, 2026-11-10:
Sure, sure, I just did not notice it.

It would be quite helpful to have it working again.

The change itself seems to be a small one(?), but the entry threshold for a programmer new in the project seems to be a bigger issue so I am a bit sceptical regarding my involvement into it.

CU