XCSoar/XCSoar #1885: iOS: Xcode not always able to launch app

WolfgangDrescher, 2025-09-16:
XCSoar versions having the problem
-------------------------------------------------

7.44-iOS

Current behavior
---------------

When building XCSoar for iOS, the app often fails to launch with the error: `The application failed to launch.`. After installing Xcode 26, the error message has become more precise: `The requested application org.xcsoar.XCSoar is not installed. Provide a valid bundle identifier.`

Expected behavior
-----------------

The iPhone should launch the freshly built XCSoar app and open it automatically with the Xcode debugger running (using the “Debug executable” option in the scheme). Before Xcode 26, however, the debugger was not running reliably. I had more success with the debugger when I built the app for the simulator beforehand. This issue is probably related to the invalid bundle identifier.

Do you have an idea how to solve the issue?
-------------------------------------------

In `build/ios.mk`, should the constant `IOS_APP_BUNDLE_IDENTIFIER` perhaps be prefixed with `org.xcsoar.`? What do you think about this, @yorickreum?

WolfgangDrescher, 2025-09-16:
Could it be a problem that the development build of this project in Xcode uses the same bundle identifier as the official App Store version? Since I'm using the TestFlight version, it's not an issue for me personally, but perhaps we should introduce a third bundle identifier (and maybe even a distinct app color)?

yorickreum, 2025-09-21:
On first glance, it looks like Xcode doesn't properly allow symlinks and `.ipa` files anymore as executables in the build scheme settings ... putting the signed .app there still works. The `IOS_APP_BUNDLE_IDENTIFIER` just has to match your development certificate and provisioning profile (if you wanna run on an actual device, not simulator). I think Xcode extracts the bundle identifier to launch from the `.app` bundle file correctly. Not sure about `.ipa` currently. Will look into how to update/harden this.